auviitkgp/kraken_3.0 #68: Launch file creation

icyflame, 2015-09-24:
- I propose the separation of this task into two parts:

1. Each package has a launch file (if it has any dependencies).

    Eg: `buoy` package will have a launch file that will run `bottomcam` package, 

2. There will be some global launch files, for specific tasks: such as:

    * `YawControl` (to run all dependencies of YawControl)
    * `BuoyTask` (which will include the particular package's launch file after running other packages)
    * `imuDvl` (to run the IMU and the DVL)

Opinions: @prudhvid @nevinvalsaraj @auviitkgp/embedded-team

prudhvid, 2015-09-28:
@auviitkgp/developers this has to be the way. The launch files will specify all the arguments needed to run a script.
Hierarchy of launch files will help a lot

icyflame, 2015-09-28:
@prudhvid Can you please give an example as to what you mean by `arguments`?

prudhvid, 2015-09-28:
By arguments I mean any files (especially parameter files) needed to be supplied as arguments for the scripts. e.g. bottomcam needs 0 as argument to open. marker task needs threshold.txt as argument. These can be specified within launch file itself.  e.g. https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/vision_stack/task_marker/launch/run_server.launch

icyflame, 2015-09-29:
@prudhvid :+1: