auviitkgp/kraken_3.0 #70: Moving to a proper logging system

icyflame, 2015-09-24:
http://wiki.ros.org/roscpp/Overview/Logging

http://wiki.ros.org/rospy/Overview/Logging

ROS has inbuilt support for Logging. We just have to use it.

We will have to port all of our code from existing `cout`, `std::cout` and `print` statements to `ROS_INFO`, `ROS_DEBUG`, etc.

It would be helpful if everyone quickly edited their projects and made these changes, and opened a PR. I would be able to merge them in fast.

**Also, I would urge whoever is writing code from here on out to strictly use these instead of using stray `cout` and `print` statements in C++ and Python, respectively.** (cc @kalyan-kumar , Barno and Priyank)

Opinions? @prudhvid @nevinvalsaraj @VishnuDuttSharma 

cc @auviitkgp/developers @auviitkgp/embedded-team

icyflame, 2015-09-26:
**update regarding log levels**

- Print all debug messages using the `rospy.logdebug` or the `ROS_DEBUG` function.
- Also, if the CLI argument `--debug` or `--verbose` is passed to the executable call statement, then initialise the node with `loglevel` as `DEBUG`, so that the `/rosout` topic will become the central place to recieve all log information.

cc @auviitkgp/developers @auviitkgp/embedded-team @auviitkgp/maintainers

icyflame, 2015-10-02:
I have abstracted the verbose tag code as a function in the `resources` package, under the `tools` namespace for CPP and under `resource.tools` for Python.

Refer to https://github.com/icyflame/kraken_3.0/commit/857f5ab9052491525dd85577a068103018bdc384 for CPP.

Refer to https://github.com/icyflame/kraken_3.0/commit/168940aac3e9099a758b3530ff23bc3d00286163 for Python.

cc @auviitkgp/developers @auviitkgp/embedded-team

prudhvid, 2015-10-02:
@icyflame :+1: nice work. And also assign this task to others. You don't do the shift all alone.
@kalyan-kumar @ManviG @pranaypratyush  you take some of the other packages and shift them to the new logging system.

icyflame, 2015-10-14:
### References for C++ and Python

- C++ - [Line number 20](https://github.com/icyflame/kraken_3.0/blob/857f5ab9052491525dd85577a068103018bdc384/control_system_stack/thruster_converter/src/thruster.cpp#L20) and [Line numbers 100-103](https://github.com/icyflame/kraken_3.0/blob/857f5ab9052491525dd85577a068103018bdc384/control_system_stack/thruster_converter/src/thruster.cpp#L100)

- Python - [Line number 29](https://github.com/icyflame/kraken_3.0/blob/168940aac3e9099a758b3530ff23bc3d00286163/control_system_stack/absolute_rpy_publisher/src/main.py#L29) and [Line number 68](https://github.com/icyflame/kraken_3.0/blob/168940aac3e9099a758b3530ff23bc3d00286163/control_system_stack/absolute_rpy_publisher/src/main.py#L68)