auviitkgp/kraken_3.0 #73: rqt_plot integration

kshitijgoel007, 2015-09-30:
To do tuning of Fuzzy controller, we tried to integrate dynamic plotting for two rostopics : 
1. /kraken/sensor/dvl/bi_x_axis
2. /kraken/sensor/imu

Issues : 
With the first one : it is showing gibberish values on x-axis. http://wiki.ros.org/Clock#Using_Simulation_Time_from_the_.2BAC8-clock_Topic

Please have a look at the above link. I think we have to find a way to use time of uwsim.

With the second one the issue is that the subtopics like "orientation x" etc. are not being shown in the rqt_plot GUI.

kshitijgoel007, 2015-09-30:
cc @prudhvid @icyflame @kalyan-kumar @ManviG

icyflame, 2015-09-30:
@kshitijgoel007 When you say:

> subtopics like "orientation x" etc. are not being shown in the rqt_plot GUI.

You mean the `orientation_x` variable in the IMU message definition? And AFAIK, I have seen that feature in `rqt_plot`. It is there in the right top corner, in the drop down box. Tell me if I am wrong.

kshitijgoel007, 2015-09-30:
Yeah, the feature is there. I'm using that feature only. But the problem is /kraken/sensor/imu/ is showing up. Not the subtopics.

Whereas in /kraken/sensor/dvl, subtopics also show up. Like : /kraken/sensor/dvl/bi_x_axis.

It seems ROS is not able to recognize subtopics of imu rostopic.

kshitijgoel007, 2015-09-30:
[image]
[image]

See these pictures. rqt_plot is able to detect dvl topic ('+' sign), but not the imu topic ('+' sign not activated)

icyflame, 2015-09-30:
@kshitijgoel007 I think that neither IMU nor DVL have "sub topics", it just has variables inside it's message. See the [IMU message file](https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/msgs_stack/kraken_msgs/msg/imuData.msg) and the [DVL message file](https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/msgs_stack/kraken_msgs/msg/dvlData.msg). They are identical except for the size of the array that they have.

Thanks for the screenshot!

Let me check if the simulator uses different message files for publishing the DVL data.

It is possible that the simulator publishes a lot more data than our physical DVL does. So, @kalyan-kumar and @prudhvid may have had to redirect only stuff that we need into the `/kraken/sensor/dvl` topic. Although, this is just a theory.
***
cc @prudhvid @kalyan-kumar @aniket11bh

icyflame, 2015-09-30:
Okay, I found the [remap code](https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/simulator_stack/thruster_remap_g500/launch/remap.launch#L10).

@prudhvid and @kalyan-kumar Did you check whether the IMU and DVL messages that we use are identical to the ones used by the simulator? I think that's the reason that we are not able to get proper data from the simulator.

cc @aniket11bh @kshitijgoel007

kshitijgoel007, 2015-09-30:
Yeah, I mistakenly called them "*subtopics*". They are variables indeed.
But we intend to use/subscribe them separately in out controller-ROS
integration code. So we need separate access to them. Like :
/kraken/sensor/imu/orientation_x.

DVL topic is well arranged.

icyflame, 2015-09-30:
@kshitijgoel007 Okay. Let's wait for Prudhvi or Kalyan to respond.

kalyan-kumar, 2015-09-30:
We didn't check if they were similar. But yesterday while discussing with Aniket that is what we have concluded.
And yes, those are just variables of the message and not subtopics.

kalyan-kumar, 2015-09-30:
Topics /g500/imu and /g500/dvl, which we have remapped to topics /kraken/sensor/imu and /kraken/sensor/dvl respectively get messages underwater_sensor_msgs/DVL and sensor_msgs/Imu. But our topics get the messages dvlData and imuData. But after remapping the messages that /kraken/sensor/imu and /kraken/sensor/dvl get are underwater_sensor_msgs/DVL and sensor_msgs/Imu. We need to see how the values are being calculated in these messages and change our calculations accordingly to get proper values in our topics.

icyflame, 2015-09-30:
@kalyan-kumar Yes, indeed. That's right.

P.S. Please use [GFM](https://guides.github.com/features/mastering-markdown/) next time onwards for better markup. I edited your comment and split up logical sections into different paragraphs.

icyflame, 2015-10-01:
This issue is the same as #74 .

It is confusing to have two issues about the same thing. I am closing this in favor of #74.

closed 2015-10-01 by icyflame