auviitkgp/kraken_3.0 #74: Issue with rostopic /kraken/sensor/imu

kshitijgoel007, 2015-09-30:
We are unable to access several subtopics explicitly in this rostopic. Whereas in /kraken/sensor/dvl we are able to access various topics explicitly : for example : /kraken/sensor/dvl/bi_x_axis, etc.

We want this functionality in this imu topic also.

prudhvid, 2015-09-30:
@kshitijgoel007 what topics do u need?

kshitijgoel007, 2015-09-30:
We need the following things as subtopics :

   1. acceleration ax : Always showing zero right now
   2. acceleration ay : Always showing zero right now
   3. acceleration az : Always showing zero right now
   4. orientation \theta x
   5. orientation \theta y
   6. orientation \theta z

We are unable to figure out following things:

   - Why are accelerations being shown zero always?
   - Why are there 4 values in "orientation" {x, y, z, w}, instead of 3. Is
   it in quaternion <https://en.wikipedia.org/wiki/Quaternion>
   representation?

prudhvid, 2015-09-30:
orientation is in quaternion. All ros communications orientation will
always be in quaternion.
And regarding ax,ay,az there's a problem with IMU corresponding to that
which we'll fix soon.

kshitijgoel007, 2015-09-30:
Ok.

Me and Aniket are working on a transformation from quaternion system to Euler angle system. It's just a transformation matrix. We can integrate such a thing in a roslaunch file right? Just matrix multiplication stuff.

Actually, quaternions are not intuitive. They are just mathematical abstractions. So, we can't directly figure out which values increase or decrease when certain moments are applied. In order to validate that the data we are getting is actually right, we need to do this.

We'll share the matrix by night.

cc @icyflame @prudhvid @kalyan-kumar

icyflame, 2015-09-30:
@kshitijgoel007 No, we won't be able to integrate them in a launch file. We would write a new node in Python and then integrate **that** node into the launch file. It would be faster if you guys could write a function that looked something like this:

```
function quaternionToEuler(x, y, z, w):
    
    # returns a list of 3 elements: [roll, pitch, yaw]
```

Would this be possible?

cc @aniket11bh

aniket11bh, 2015-09-30:
I think we don't need to write any extra function because they are used widely.
http://wiki.ros.org/geometry/RotationMethods

We can use any of the methods from the above link.
transformations.py would be good for this.

icyflame, 2015-09-30:
@aniket11bh Awesome! :+1: Using well-written :package: 's is best.

Can you and @kshitijgoel007 put together a PR on the `thruster_remap_g500` package?

kshitijgoel007, 2015-09-30:
> @aniket11bh <https://github.com/aniket11bh> Awesome! [image: :+1:] Using
> well-written [image: :package:] 's is best.

Damn. I forgot to google it. I started reading Fossen :P

aniket11bh, 2015-09-30:
In my git hub repo i've added a subscriber to get the data from
kraken/sensor/imu and converted into rpy form.
It's working fine. Yaw is changing perfectly for corresponding change in
vehicle yaw
- But still control.py is not working
- yaw is coming negative and positive as well so, we need to get the
absolute data of this.

Sent with MailTrack
<https://mailtrack.io/install?source=signature&lang=en&referral=aniket2701bhushan@gmail.com&idSignature=22>

aniket11bh, 2015-10-01:
This is the what i'm getting after running control.py , it seems like control.py is not publishing the thruster output data anywhere. @icyflame  and @kalyan-kumar , you guys told me that we don't need to change anything inside the file control.py for running the thrusters from the simulator, but it seems like this is not the case. And also seabotix.py is giving out the error (obviously it would because arduino is not connected).
So, reply asap what to do now ? Who is fixing it?
[image]

icyflame, 2015-10-01:
@aniket11bh As discussed with you, I have opened #75 .

Once #75 is fixed by @kalyan-kumar , we will be able to run `control.py` and have it working as expected.

cc @kshitijgoel007 @prudhvid 

P.S. I have closed #73 in favour of this issue. [This comment](https://github.com/auviitkgp/kraken_3.0/issues/73#issuecomment-144414568) by @kalyan-kumar probably summarises the problem we were facing.

PR #76 (2015-10-01): Add missing remap to make sure IMU and DVL data is extracted properly
- Data from the simulator is in the messages [sensor_msgs/Imu]() and [underwater_sensor_msgs/DVL]()

- Me and @aniket11bh have converted this message into the `kraken_msgs/imuData` and `kraken_msgs/dvlData` message formats.

- And then published these formats on the IMU and DVL topics.

FIXES #74 
FIXES #75 

TODO:

- Figure out why the `linear_acceleration` and `depth` values from the simulator IMU and DVL are not as expected
- Improve `pose_server_python`. See #77.

We have already tested this code, and the `pose_server_py [... cut]
commit d0cd76de (2015-10-01): Merge pull request #76 from icyflame/convert-imu-data-from-simulator

- Add missing remap to make sure IMU and DVL data is extracted properly.
  - By converting the message types used by g500 to message types used by kraken_3.0 packages.
- Remap the thruster data sent by control nodes directly.
- Remove the necessity of running a `thruster_force_converter` node.
- Code has been tested with the yawControl node, and the architecture is setup properly. Problems in `control.py` hamper testing.
- Code has been tested with `pose_server_python`, it works!

FIXES #74 
FIXES #75
closed 2015-10-01 by icyflame