auviitkgp/kraken_3.0 #75: One remap is missing, due to which `control.py` does not work.

icyflame, 2015-10-01:
The topic

```
/kraken/control/thruster6pid
```

has to be remapped to

```
/kraken/physics_model/forceData6Thruster
```

and the corresponding four thruster model topics.

References: [Comment by @prudhvid ](https://github.com/auviitkgp/kraken_3.0/pull/72#issuecomment-143903287)

cc @kalyan-kumar @prudhvid @kshitijgoel007 @aniket11bh

icyflame, 2015-10-01:
@kalyan-kumar My bad. I was talking about [this `control.py`](https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/sensor_stack/seabotix/src/control.py) file.

aniket11bh, 2015-10-01:
There is some problem in thruster_force_converter, because 
- /Control node is giving a good data for yaw movement. while /converter is giving a constant output 0 corresponding to the same thrusters as of from /Control node (Check in  rqt graph and data as attached)
-Left part of terminal is from the control.py while right part is from thruster_force_converter
[image]

[image]

icyflame, 2015-10-01:
@aniket11bh 

As far as I can see, the rqt_graph is exactly as expected by us. This is what is happenning:

- IMU value is being taken by the Control node
- Some processing happens and the thruster force values are published to `/kraken/control/thruster4pid`
- This is remapped to `/kraken/physics_model/forceData4Thruster` which is the topic that the simulator takes thruster input from.

The feedback loop is complete. And the basic architecture is setup.

Regarding the data, I am not able to understand what topics each of the terminal screens above correspond to. `thruster_force_converter` is a package, and not a topic.

cc @kalyan-kumar

aniket11bh, 2015-10-01:
@icyflame 
- '/converter ' is a node which corresponds to 'main.py' in 'thruster_force_converter' package.
- Left part of the terminal screen belongs to the data as published by 'control.py' on '/kraken/control/thruster6pid'
- Right part of the screen belongs to the data as published by the node '/converter' on '/kraken/physics_model/forceData6Thruster' .
- '/converter' which is a node generated by .cpp files (either by 'main.cpp' or by 'ForceConverter.cpp')in 'thruster_force_converter' package

PR #76 (2015-10-01): Add missing remap to make sure IMU and DVL data is extracted properly
- Data from the simulator is in the messages [sensor_msgs/Imu]() and [underwater_sensor_msgs/DVL]()

- Me and @aniket11bh have converted this message into the `kraken_msgs/imuData` and `kraken_msgs/dvlData` message formats.

- And then published these formats on the IMU and DVL topics.

FIXES #74 
FIXES #75 

TODO:

- Figure out why the `linear_acceleration` and `depth` values from the simulator IMU and DVL are not as expected
- Improve `pose_server_python`. See #77.

We have already tested this code, and the `pose_server_py [... cut]
commit d0cd76de (2015-10-01): Merge pull request #76 from icyflame/convert-imu-data-from-simulator

- Add missing remap to make sure IMU and DVL data is extracted properly.
  - By converting the message types used by g500 to message types used by kraken_3.0 packages.
- Remap the thruster data sent by control nodes directly.
- Remove the necessity of running a `thruster_force_converter` node.
- Code has been tested with the yawControl node, and the architecture is setup properly. Problems in `control.py` hamper testing.
- Code has been tested with `pose_server_python`, it works!

FIXES #74 
FIXES #75
closed 2015-10-01 by icyflame