auviitkgp/kraken_3.0 #82: gitignore is set to ignore these message files

icyflame, 2015-10-05:
Source: https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/.gitignore#L7

```shell
**/msg/*Action.msg
**/msg/*ActionFeedback.msg
**/msg/*ActionGoal.msg
**/msg/*ActionResult.msg
**/msg/*Feedback.msg
**/msg/*Goal.msg
**/msg/*Result.msg
**/msg/_*.py
```

We found this problem when @aniket11bh tried to add a message file called `controllerResult.msg` inside `kraken_msgs/msg/` folder.

@nevinvalsaraj :  

- Why had you added these lines?
- Don't we think we should make them more specific?

cc @prudhvid @auviitkgp/developers

prudhvid, 2015-10-05:
@icyflame this was done because when we write an action file for ActionLibServer, it generates the following messages http://wiki.ros.org/actionlib (check 5.3). So, I advised to add the following lines to gitignore to @nevinvalsaraj . So please ask aniket to change the name of message and put up a PR again

icyflame, 2015-10-05:
​As a matter of fact, we added the file using `git add filename -f` which
force adds a file. That's fine for most cases, but having a related file
name is important. I will look into making the line more specific (as in
something else rather than just `**/msg/*.msg` which is equivalent to any
`.msg` file inside the `msg` folder of any sub-directory.​

closed 2015-10-05 by icyflame