auviitkgp/kraken_3.0 #84: Change all the parameters hard-coded across the repository to ROS parameter server

icyflame, 2015-10-28:
The issue was first raised [in this comment](https://github.com/auviitkgp/kraken_3.0/pull/83/files#r43304475)

References:

1. [Parameter Server](http://wiki.ros.org/Parameter%20Server)
2. [rosparam](http://wiki.ros.org/rosparam)

cc @auviitkgp/developers @auviitkgp/embedded-team @auviitkgp/maintainers

icyflame, 2015-12-13:
Okay, I am moving this up in priority because this is way better than a lot of other approaches to do this.
For example, we can remove a lot of the pub-sub stuff that we have created by simply creating a `rosparam` for it and then, getting that `rosparam`.
One use-case may be for thrusters, where sending the data through nodes can be easily done instead as setting a rosparam, and this has the advantage that we can check if the parameter is null.
For constants, and other things that are generally tuned, this is **certainly the way to go**.
Further, it is a live server, so, you don't need to restart the node, setting the parameter anywhere will immediately affect all nodes using this.
Also, implementing this (in both C++ and Python) is a breeze because of the easy API provided by ROS.

**C++ example:** [thruster_converter module](https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/control_system_stack/thruster_converter/src/thruster.cpp#L120-L131)

**Python example:** [PID Control module](https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/sensor_stack/seabotix/src/pidControl.py#L119-L122)

Attn: @prudhvid @kalyan-kumar @ghostwriternr @nevinvalsaraj 

cc @auviitkgp/developers

prudhvid, 2015-12-13:
Fantastic this is THE WAY to go :)

pranaypratyush, 2016-01-09:
Why is "/ros_rate" parameter 'string' instead of 'int' ? And where are we creating this parameter ? (in some '.launch' file ?

icyflame, 2016-01-09:
@pranaypratyush Where is the option to declare this as float? I used the default commands, and it turned out to be string. Wasn't that big a deal, so I left it. Both Python and C++ can easily convert. 

Of course, if there is a data type field, then we should use `integer` for `ROS_RATE`

pranaypratyush, 2016-01-10:
Yes we can declare the parameters as `integers`  (32 bit actually). But this still doesn't answer my question as to where is the parameter declared ?

icyflame, 2016-01-10:
Check the first two links. The values are stored and fetched from the parameter server. I just missed the parameter type section in [the first doc](http://wiki.ros.org/Parameter%20Server)

I will assign a new issue to you. You must change all the present occurrences of ROS Rate to the integer and also create the required YAML file.

pranaypratyush, 2016-01-10:
Peace. Do it.

icyflame, 2016-01-10:
Done. Check #98. I think one week is enough to get this done?