auviitkgp/kraken_3.0 #87: Handle the case when goal is preempted by client inside the server in setyaw

icyflame, 2015-10-30:
- Working on @aniket11bh 's machine
- Not working on mine, although we are using the same version of the code, and the stack is similar.

It would really help if someone else can test this on their machine as well. @kshitijgoel007  @kalyan-kumar @ManviG Can you guys check and tell me what's really going on?

Opinions? @prudhvid @nevinvalsaraj 

cc @auviitkgp/developers

kshitijgoel007, 2015-10-30:
Mine also not working.

kalyan-kumar, 2015-10-30:
Tested on my machine and it was working properly. The goal was preempted by the client.

icyflame, 2015-11-23:
@aniket11bh 

I just tested this again, and it worked on my machine, but the client said **You pressed Ctrl+C**, and that was it. It didn't say that the goal was preempted by it. A message saying that might be helpful. Can you add that to the present setyaw client code and submit a PR?

[image]

@kalyan-kumar Is that what you saw as well?

icyflame, 2015-12-10:
Works fine now. Closing this issue!

closed 2015-12-10 by icyflame