auviitkgp/kraken_3.0 #89: Add a list of the packages to be installed in order for yawControl to work in the readme

icyflame, 2015-11-04:
- Add a `bootstrap.sh` bash script?
- Need to install `python-scipy` and `scikit-fuzzy` and required dependencies.

@aniket11bh @kalyan-kumar Anything that I am missing?

kalyan-kumar, 2015-11-04:
Yes. That's all we installed on my machine to run the package.

icyflame, 2015-11-07:
@kalyan-kumar Can you go through your Bash `history` and post the command that you had run to install everything.

```sh
$ history | grep -i scipy
```

should work. (It greps through the bash history and outputs the command, if found) :smile:

kalyan-kumar, 2015-11-08:
<addr> sudo apt-get install python-numpy python-scipy python-matplotlib ipython ipython-notebook python-pandas python-sympy python-nose

icyflame, 2015-11-12:
@kalyan-kumar @aniket11bh @kshitijgoel007 Added to the new readme, check it out.

commit 09127a2d (2015-11-12): Merge branch 'add-installation-commands-to-readme' into indigo-devel

FIXES #89
closed 2015-11-12 by icyflame