auviitkgp/kraken_3.0 #95: `offset` not defined inside thruster_converter

icyflame, 2015-12-10:
It is referenced on [line 41](https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/control_system_stack/thruster_converter/src/thruster.cpp#L41)

I have defined it as `0x80` on [line 25](https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/control_system_stack/thruster_converter/src/thruster.cpp#L41) **to ensure that the code compiles without any errors**.

What should be the value of this variable?

cc @VishnuDuttSharma

aniket11bh, 2015-12-10:
It should be zero speed as per thrusters. So, either of offsetF or offsetB will work.

 - Also in [line 48] (https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/control_system_stack/thruster_converter/src/thruster.cpp#L48) and [line 82](https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/control_system_stack/thruster_converter/src/thruster.cpp#L82), it should be `0x19-0x7F` instead of `0x19-0x80`, considering the symmetry of speeds of the thruster on either side.

- In lines 47 and 48 `?(0xE6-0x80)` should be replaced with `?0xE6` and similarly `?(0x19-0x80)` by `?0x19` like line numbers 81 and 82

icyflame, 2015-12-10:
​Okay, since there are so many changes, can you open a PR on the latest
indigo-devel branch?​

icyflame, 2015-12-13:
ping @aniket11bh

icyflame, 2016-01-10:
ping @aniket11bh

aniket11bh, 2016-01-10:
The code is already complete. It is on vehicle. I'll get the code and create a PR

icyflame, 2016-01-21:
Has this been fixed in the commit that was merged as a result of #102  ?

kshitijgoel007, 2017-01-07:
@aniket11bh What is the status of the PR?