auviitkgp/kraken_3.0 #96: DOCS: Add a page about the network settings to be done on the vehicle's motherboard for SSH to work

icyflame, 2015-12-13:
- Also, add photos.

pranaypratyush, 2016-01-08:
Wait tell I get start testing the bot. I was planning on installing a DHCP server on the bot to make connecting a little bit easier. I would create the docs myself after it's done. Also suggest other enhancements to make connecting easier (and also to allow us to connect the bot to the internet without much effort).

icyflame, 2016-01-09:
@pranaypratyush DHCP servers don't need to be installed. They are bundled with Ubuntu. We can't use DHCP, it would make connecting harder. Static IP ssh-ing is easier because we know the IP. We can't find the IP of a DHCP computer on the network. Or can we?

And this issue is my work, because I have the photos. I will create a skeleton page, you can improve on it.

icyflame, 2016-01-10:
Added page with screenshots: https://github.com/auviitkgp/kraken_3.0/wiki/Network-settings-to-be-made-on-the-vehicle

cc @pranaypratyush @prudhvid 

@pranaypratyush If you want to add more details, feel free. Wiki can be edited by anyone.

closed 2016-01-10 by icyflame

pranaypratyush, 2016-01-10:
I think you got it wrong. DHCP won't decide the ip of the bot. We will set that as some default value ( and  maybe give it a default NETBios domain name ). The DHCP server will assign IPs to the clients (our lappy(s)). After this is done we won't require any configuration change on our lappy to ssh. Also we would have a domain name to ssh to instead of remembering an ip.

icyflame, 2016-01-10:
Hmm, okay. But I still don't see any significant advantage. IP addresses are not a stretch anyway, they can be automated with one simple shell alias. And it's a onetime configuration, and no installation is required. So, I think our present option is better, unless we see some significant advantage in installing DHCP server on the vehicle.

@prudhvid and @nevinvalsaraj What are your takes?

pranaypratyush, 2016-01-10:
Yes of course. But being engineers I think the whole purpose of our life is automation (and taking other people's jobs). Plus your methods ( even the ssh command ) would still work after I have configured. Of course it's a non-essential feature but why not do it if we can ( would take like 30 min to test and won't have to touch it again. also it won't interfere with anything else ).

icyflame, 2016-01-10:
@pranaypratyush I really don't get it. Why should we do it when there's no value addition? It would install another package, which is completely unnecessary.

> Just coz we are engineers

does not cut it.

[image]

P.S. Interesting discussion though.

cc @auviitkgp/developers

pranaypratyush, 2016-01-10:
LOL.

nevinvalsaraj, 2016-01-18:
@pranaypratyush I'm all for automating stuff, but I think a whole DHCP server for a one node network is overkill. After all, engineers prioritize optimization.
@icyflame Maybe we can have a shell snippet with the required aliases and so forth that can be sourced to `.shellrc` by anyone wanting to configure an SSH connection.

icyflame, 2016-01-18:
@nevinvalsaraj Good idea, I have been thinking about that. I have already added `rsync` commands for syncing local code onto vehicle and vehicle code onto machine, but they are in the [dotfiles](http://github.com/auviitkgp/dotfiles) repository. I will isolate this stuff into a `.shellrc` file.