auviitkgp/kraken_3.0 #98: Change the ROS_RATE everywhere to a 32-bit integer ROSParam

icyflame, 2016-01-10:
related to #84 .

pranaypratyush, 2016-01-12:
I am using double. It seems it would make things easier.

pranaypratyush, 2016-01-12:
The default rate differs throughout the repo. It's mostly 8 or 10 but also 5 ( nqDVL.py). Using the same rate everywhere seems safe to me, but just wanted to make sure if it's actually okay.

icyflame, 2016-01-12:
Yes, @pranaypratyush It should be the same. Especially in the modules that are connected.

For now, link them all to one ROSParam. We may have to restructure some code for this to work. For now, just change all ros_rate instances to the same value and then, put this value in a YAML file that can be dumped into ROSParam server. I hope that is what you are doing? 

We will have a separate discussion about the rates, once you open a PR and we will have a diff at hand.

cc @prudhvid

pranaypratyush, 2016-01-12:
Also we can get rid of a separate yaml file. Just  put 
`<rosparam param="/ros_rate">10.0</rosparam>`
 in a roslaunch file. Seems neat to me.

icyflame, 2016-01-12:
@pranaypratyush No, we **must** use a YAML file. You must note that there will be a whole lot of parameters (We also have to move Fuzzy Control parameters, PID control parameters, etc etc) into the rosparam server. 

So, for that the easiest way to start is to use a YAML file. Instead of having to put that line again and again in each launch file, or one huge launch file, we simply have a YAML file which can be easily dumped.

pranaypratyush, 2016-01-12:
Ok I am creating a new folder @ `/kraken_3.0/Scripts/launch/yaml` and creating a yaml file. Also I would be editing all `.launch` files to load the `yaml` file.
Nod if you agree.

icyflame, 2016-01-12:
:+1: 

I am not sure about the path now, but we can trivially change it later, if required

icyflame, 2016-05-22:
@pranaypratyush Have this on your todo list for testing once the vehicle is assembled and ready to be deployed in the water.

pranaypratyush, 2016-05-22:
Sure

closed 2017-08-07 by icyflame