auviitkgp/kraken_3.0 #100: Change the IMU message type to Sensor_msgs/Imu

icyflame, 2016-01-13:
- Will standardize the message type for IMU messages that we are using.
- Will help out a lot when using standard ROS packages / plugins (such as ros-control-center referred to [here](https://github.com/auviitkgp/auv-core-gui/pull/1#issuecomment-170467761))

## Migration Procedure

- We edit the `sparton` package.
- We publish the new message type on a second IMU topic (`/kraken/sensor/imu_2`) **(First pull request)**
- We start editing the modules using IMU data. (One pull request per module)
- We can slowly phase out the old topic once we are sure no other packages depend on it.

@prudhvid @nevinvalsaraj Opinions on the above procedure?

cc @auviitkgp/developers

nevinvalsaraj, 2016-01-18:
Rather than making a second IMU topic with the desired data format, I'd say replace the original. We have way too many original topic + converter package + modified topic clusters written as it is. Seems like a waste of transport and processing power.

icyflame, 2016-01-18:
@nevinvalsaraj I am afraid that completely replacing the old topic will cause unnecessary issues with old packages. And we should rather go for this kind of a migration procedure, with a pre-set date for the removal of the old topic entirely. (say 1 month from now, leading to around 4-5 pool testing days)

nevinvalsaraj, 2016-01-18:
@icyflame That's fine. We can phase out the old topic, but my point being that this needs to be done rather than keeping both of them side-by-side. :+1:

icyflame, 2016-01-19:
Yes, @nevinvalsaraj . Actually, this also needs to be done for some older packages. Me and @prudhvid had decided that we would move all such packages into an `obsolete/` folder. (Such as `nqDvl` CPP package, trax IMU, etc).

commit 4cbad65d (2016-08-28): Merge branch 'athityakumar-sparton-new' into indigo-devel

FIXES #112
FIXES #100

Not tested on the Vehicle yet.

Shortlog:

Parth Verma (4):
      Changed cov_mat to ROS_Param type
      Create ros_sparton.yaml
      Update imuDvl.launch
      Update sparton.py

athityakumar (11):
      Partially modified spartion. Cov matrices & quaternion remaining.
      Set cov matrices to 0.Conversion of rpy to quaternion remaining.
      Changed Vector to Vector3. Started on quaternion.
      Added (mx,my,mz) & temp in magnetoTemp.msg. Calculated Quaternion.
      Used default Imu message. Added topic i [... cut]
closed 2016-08-28 by icyflame