auviitkgp/kraken_3.0 #101: [HOUSEKEEPING] Move unused packages to the obsolete folder

icyflame, 2016-01-19:
Currently, the list of packages is:

- [NqDVL CPP](https://github.com/auviitkgp/kraken_3.0/tree/indigo-devel/sensor_stack/NqDVL)
- [Trax](https://github.com/auviitkgp/kraken_3.0/tree/indigo-devel/sensor_stack/trax)
- [Tracks IMU](https://github.com/auviitkgp/kraken_3.0/tree/indigo-devel/sensor_stack/tracks_imu)
- [Depth sensor?](https://github.com/auviitkgp/kraken_3.0/tree/indigo-devel/sensor_stack/depth_sensor) (cc @aniket11bh @kshitijgoel007 I haven't really ever used this package. Is this useful **right now**? If this is a planned enhancement, we can move this over to obsolete, and then move it back here when necessary.)

cc @prudhvid @nevinvalsaraj Any other packages that you feel are not useful anymore? Probably in the simulator_stack?

fyi @auviitkgp/developers

aniket11bh, 2016-01-19:
Depth sensor package will be used in future when our sensor will be ready.
ᐧ

icyflame, 2016-01-21:
Fixed in 8ff7890921f14ea5fd17c9a667bf55830dfbcdcd

closed 2016-01-21 by icyflame