auviitkgp/kraken_3.0 #106: Check 'Thruster converter package'

aniket11bh, 2016-03-20:


icyflame, 2016-03-20:
@aniket11bh Simulator? Which launch file did you run? And WDYM:

> not working in our laptops

aniket11bh, 2016-03-20:
When you make the 'thruster_converter' package.

icyflame, 2016-03-20:
@aniket11bh Exact command and the error you get please

aniket11bh, 2016-03-20:
`rosmake thruster_converter`

aniket11bh, 2016-03-20:
Check this one if above one looks incomplete

On Sun, Mar 20, 2016 at 4:53 PM, Aniket Bhushan <aniket2701bhushan@gmail.com

> `rosmake thruster_converter`
>
> On Sun, Mar 20, 2016 at 3:56 PM, Siddharth Kannan <
>
>> @aniket11bh <https://github.com/aniket11bh> Exact command and the error
>> you get please
>>
>>
>
>
>
> Aniket Bhushan
> 3rd year Undergraduate ,
> Manufacturing Science and Engineering
> IIT Kharagpur

icyflame, 2016-03-20:
There seems to be a `printf` statement that is causing a problem, I have removed and pushed to `indigo-devel`.

- Update your repository to latest `indigo-devel` branch
- `cd` into the `thruster_converter` package
- `rosmake`

I just built that package and it worked for me.

If that doesn't work, **paste the COMPLETE command and the error you get when you run that command here**.

icyflame, 2016-03-20:
Fixed in https://github.com/auviitkgp/kraken_3.0/commit/90c78e52d6e2913e4adc677640628bcacea6bb77

closed 2016-03-20 by icyflame