auviitkgp/kraken_3.0 #107: 'Changes in thruster mapping in `control_system_stack/setyaw/src/main.py`

aniket11bh, 2016-03-21:
Why have you changed the thruster mappings ?

[image]

icyflame, 2016-03-21:
@aniket11bh Apparently, it made PID work. I think this was during december testing, is it not working now? I remember thruster mappings were repeatedly changed during winter. @kalyan-kumar Remember anything from then?

aniket11bh, 2016-03-21:
The changed mapping is not suitable with the simulators, the previous one
is.

icyflame, 2016-03-21:
Hmm, that is a problem. Why not send a patch that takes a CLI argument to decide the thruster mapping? (Default will be the vehicle one of course, but just in case we want to run the simulator, we can pass in something like `simulator` after `./bin/exec-name`.)