auviitkgp/kraken_3.0 #109: Error in blob package

pranaypratyush, 2016-05-22:
When i try to build the `blob` package with `clang++` , I get this error.
                                                                                                                       ~~~~^
4 warnings generated.
[ 50%] Building CXX object CMakeFiles/blob.dir/src/ComponentLabeling.cpp.o
In file included from /home/pranay/ros_ws/kraken_3.0/vision_stack/blob/src/ComponentLabeling.cpp:2:
In file included from /home/pranay/ros_ws/kraken_3.0/vision_stack/blob/include/blob/ComponentLabeling.h:6:
/home/pranay/ros_ws/kraken_3.0/vision_stack/blob/include/blob/blob.h:16:9: warning: unknown pragma ignored [-Wunknown-pragmas]
#pragma warning( disable : 4786 )
        ^
/home/pranay/ros_ws/kraken_3.0/vision_stack/blob/src/ComponentLabeling.cpp:6:3: error: non-aggregate type 'const CvPoint' cannot be
      initialized with an initializer list
{ {1, 0}, {1, -1}, {0, -1}, {-1, -1}, {-1, 0}, {-1, 1}, {0, 1}, {1, 1} };
  ^~~~~~
/home/pranay/ros_ws/kraken_3.0/vision_stack/blob/src/ComponentLabeling.cpp:6:11: error: non-aggregate type 'const CvPoint' cannot be
      initialized with an initializer list
{ {1, 0}, {1, -1}, {0, -1}, {-1, -1}, {-1, 0}, {-1, 1}, {0, 1}, {1, 1} };
          ^~~~~~~
/home/pranay/ros_ws/kraken_3.0/vision_stack/blob/src/ComponentLabeling.cpp:6:20: error: non-aggregate type 'const CvPoint' cannot be
      initialized with an initializer list
{ {1, 0}, {1, -1}, {0, -1}, {-1, -1}, {-1, 0}, {-1, 1}, {0, 1}, {1, 1} };
                   ^~~~~~~
/home/pranay/ros_ws/kraken_3.0/vision_stack/blob/src/ComponentLabeling.cpp:6:29: error: non-aggregate type 'const CvPoint' cannot be
      initialized with an initializer list
{ {1, 0}, {1, -1}, {0, -1}, {-1, -1}, {-1, 0}, {-1, 1}, {0, 1}, {1, 1} };
                            ^~~~~~~~
/home/pranay/ros_ws/kraken_3.0/vision_stack/blob/src/ComponentLabeling.cpp:6:39: error: non-aggregate type 'const CvPoint' cannot be
      initialized with an initializer list
{ {1, 0}, {1, -1}, {0, -1}, {-1, -1}, {-1, 0}, {-1, 1}, {0, 1}, {1, 1} };
                                      ^~~~~~~
/home/pranay/ros_ws/kraken_3.0/vision_stack/blob/src/ComponentLabeling.cpp:6:48: error: non-aggregate type 'const CvPoint' cannot be
      initialized with an initializer list
{ {1, 0}, {1, -1}, {0, -1}, {-1, -1}, {-1, 0}, {-1, 1}, {0, 1}, {1, 1} };
                                               ^~~~~~~
/home/pranay/ros_ws/kraken_3.0/vision_stack/blob/src/ComponentLabeling.cpp:6:57: error: non-aggregate type 'const CvPoint' cannot be
      initialized with an initializer list
{ {1, 0}, {1, -1}, {0, -1}, {-1, -1}, {-1, 0}, {-1, 1}, {0, 1}, {1, 1} };
                                                        ^~~~~~
/home/pranay/ros_ws/kraken_3.0/vision_stack/blob/src/ComponentLabeling.cpp:6:65: error: non-aggregate type 'const CvPoint' cannot be
      initialized with an initializer list
{ {1, 0}, {1, -1}, {0, -1}, {-1, -1}, {-1, 0}, {-1, 1}, {0, 1}, {1, 1} };
                                                                ^~~~~~

pranaypratyush, 2016-05-22:
Note that  i get an error if  I try to build with `gcc-5`

[100%] Linking CXX executable bin/testblob
CMakeFiles/testblob.dir/src/testblob.cpp.o: In function `main':
/home/pranay/ros_ws/kraken_3.0/vision_stack/blob/src/testblob.cpp:9: undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
collect2: error: ld returned 1 exit status

icyflame, 2016-05-22:
@pranaypratyush You should use `make` or `rosmake`. I don't think `clang` or `gcc` are the way to go.

icyflame, 2016-05-22:
@pranaypratyush If it worked with `rosmake` (if this is the first time) or with `make` or `make -j4`, this issue can be closed. Report back here, once you have tried the command.

pranaypratyush, 2016-05-22:
-_- 
Of course I was using `make` and `rosmake`. I just set their variables to use `clang` / `gcc-5` as compiler. Maybe `rosmake` works on ubuntu 14.04 but it doesn't on ubuntu 16.04. Maybe I need to rollback my version of gcc. But still is that a good thing ?
Would try with `gcc 4.8`.

icyflame, 2016-05-23:
We have used `make` and `rosmake` only with gcc till now. Better not get fancy with other compilers, and stick to what we know. Closing this now.

Also, wasn't it that ROS indigo isn't supported on 16.04? @kalyan-kumar had to forcefully install Jade, if I remember right.

closed 2016-05-23 by icyflame

pranaypratyush, 2016-05-23:
So I had to use `gcc-5.3.1` and `g++-5.3.1` along with clang in order to build this with `ros-kinetic` on `ubuntu-16.04 (xenial)` . Would be good to make a note so as to inform others who face similar problem.