auviitkgp/kraken_3.0 #140: Sparton IMU

nerellasureshkumar7, 2019-08-10:
Hi Guys, 
     I have seen that you people are using Sparton IMU for your bot. I have contacted one of your senior who(Atithya Kumar) worked on the AUV. I have some doubts in integrating the Sparton IMU sensor can you please help me out.

icyflame, 2019-08-12:
@nerellasureshkumar7 Hi, sure. We use the Sparton IMU through a serial port connection. The [code][1] for this uses the `serial` communication library in Python along with the `roslib` library to publish this information on a [ROS topic][2].

Is there anything else that you want to know about specifically? Feel free to write your questions in this issue and someone will help you with them. 👍 

[1]: https://github.com/auviitkgp/kraken_3.0/blob/indigo-devel/sensor_stack/sparton/src/sparton.py
[2]: https://github.com/auviitkgp/kraken_3.0/blob/4bbf2450b0df7405035bc88966d9b8e2244e5c74/resources/src/topicHeader.cpp#L51-L52

nerellasureshkumar7, 2019-08-12:
@icyflame Hi, Thanks for your response. I'm using this [https://github.com/udmamrl/SpartonCompassIMU](url) repository for getting the IMU values via serial port. 
1. As far i've seen the IMU default go with NED but as per ros standards(REP 103) we need ENU. After i'm converting the values into ENU the Pitch and Roll are inverted.
2. Can you suggest me the mounting position of IMU on the robot?  I placed the IMU horizontal to body frame but the acceleration values (0,0,-9.8) which should be (0,0,9.8).
    I feel that i'm missing something can you guys help me out.

icyflame, 2019-08-13:
@aniket11bh can you help with the placement of the IMU on the robot?

nerellasureshkumar7, 2019-08-18:
@icyflame Hi, could you people help me with Sparton IMU problem.

icyflame, 2019-08-21:
@nerellasureshkumar7 Hey, I graduated last year; I was working on AUV a couple years before that. So, my memory about the exact placement is slightly hazy. I will try to get someone else who is actively involved in the group to look at your issue. :slightly_smiling_face:

closed 2019-08-29 by nerellasureshkumar7